sandeepmistry/arduino-nRF5 #431: nRF52840 MDK USB Dongle support

mischief, 2020-10-23:
hi, i have one of these: https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/

unfortunately programming it with nordic's SDK is a real pain and i'm not a fan.

would it be possible to support this board in this repo? if so, how can that be done? i am willing to try changes myself if given pointers or donate a board.

Antti, 2020-11-08:
You need to define the board in `boards.txt` file and define a variant in `variants/_variant_name_/` folder (see other variants). Variants are pretty simple, you just define a pin mapping (nRF52840 pin numbers to arduino pin numbers) p2 -> D0, p3 -> D1, etc. And some aliases for common pins (leds, buttons). And that's pretty much it.

jandrassy, 2020-12-05:
https://github.com/sandeepmistry/arduino-nRF5/issues/395