sandeepmistry/arduino-nRF5 #440: setPins undeclared ?

Entretoize, 2020-11-30:
I read that it is possible to set the pins for I2C with the setPins method however when I try to do `Wire.setPins(1,2)` the compiler returns an error because it's not defined.
I don't see how to know which version is used nor how to update.

pvonmoradi, 2021-01-04:
Similar issue for Serial:  
```bash
src/main.cpp: In function 'void setup()':
src/main.cpp:6:12: error: 'class Uart' has no member named 'setPins'
     Serial.setPins(PIN_SERIAL_RX, PIN_SERIAL_TX);
```

```cpp
#include "Arduino.h"
/* #include "Uart.h" */
#include "variant.h"

void setup() {
    Serial.setPins(PIN_SERIAL_RX, PIN_SERIAL_TX);
    Serial.begin(115200);
    Serial.println("Hi from nRF52832!");
}

```

KEranda, 2021-01-25:
I had the same question a while ago and managed to find a solution after reading another issue/merge request (#213 and #221) submitted for this core. You can read this for more info but in general: 

If you're using the nRF52DK the I2C pins are 'hardcoded' by Sandeep's code to be P0.26 (SDA) and P0.27 (SCL) so all you need to do is connect you sensor to those pins and call  "Wire.begin();"

If you have selected a "Generic nRF52"board in the tools menu of the Arduino IDE you can then call "Wire.setPins(SDA, SCL);" with the pins you want to use before you call "Wire.begin();"

Hopefully it helps.

Entretoize, 2021-01-25:
Thank you I will check.

jLynx, 2021-05-11:
This solves the issue I had with Serial. Thanks @KEranda