sandeepmistry/arduino-nRF5 #441: Is nRF52805 supported? If not, how can I help?

nikolac, 2020-12-05:
According to this [comparison table](https://infocenter.nordicsemi.com/topic/struct_nrf52/struct/nrf52.html?cp=4), it looks like the nRF52805 has a subset of the functionality of the other nRF52 modules. 

I'm curious about the process of adding support for new nRF52 versions. I'm hoping it's as simple as conditionally including chunks of functionality based on the chip version.

Once working here, I plan to work on getting the MySensors library updated. 

I've ordered a couple of EBYTE NRF52805 E104-BT5005A modules and made a [breakout [oshpark]](https://oshpark.com/shared_projects/bDIdX1JU).

Ebyte Product: https://www.aliexpress.com/item/1005001709688444.html
Ebyte Datasheet [direct pdf download]: http://www.ebyte.com/en/pdf-down.aspx?id=2156

Though I'm just a hobbyist, I'll gladly take a shot at getting this going, just need a little guidance.

sandeepmistry, 2020-12-05:
Hey @nikolac,

You'd probably need a changes like this for a new variant for this:
- https://github.com/sandeepmistry/arduino-nRF5/commit/81429bf20c8f26df181e6e89c3612ccfa6a9bcfc
or 
- https://github.com/sandeepmistry/arduino-nRF5/commit/4c9101dc1677caf94bb5c2ed3af5a59bd7a08d36

nikolac, 2020-12-05:
Seems relatively straight forward. Once I get my breakouts, I'll give it a shot. 

Thanks for the help.

nikolac, 2020-12-19:
Submitted a pull request: #442

closed 2020-12-19 by nikolac