sandeepmistry/arduino-nRF5 #445: attachInterrupt with pin "LOGO" doesn't work (Micro:bit V2)

Polyphe, 2021-01-17:
Hello,
with the code here below  "attachInterrupt(digitalPinToInterrupt(LOGO), doIntLOGO, FALLING);" that didn't work, with 
attachInterrupt(digitalPinToInterrupt(BUTTONA), doIntLOGO, FALLING); that worked (???). Why ?

`// Pinout Micro:bit V2
#define BUTTONA   (5) 
#define BUTTONB   (11)

#define LOGO      (26)
#define RUN_MIC   (28)

int toggle=1;

void gpio_init(void)
{
  pinMode(BUTTONA, INPUT);  
  pinMode(BUTTONB, INPUT);
  pinMode(LOGO, INPUT);
  
  digitalWrite(RUN_MIC, LOW);
  pinMode(RUN_MIC, OUTPUT);
}

// initialise interrupt
//
void int_init(){
  attachInterrupt(digitalPinToInterrupt(LOGO), doIntLOGO, FALLING);
}

// LOGO handler (input select)
//
void doIntLOGO(){
  if(toggle<0){
    digitalWrite(RUN_MIC, LOW); // disable microphone
    toggle=1;
  }else{
    digitalWrite(RUN_MIC, HIGH); // enable microphone
    toggle=-1;
  }
}

void setup() {
  // put your setup code here, to run once:
  gpio_init();
  int_init();
}

void loop() {
  // put your main code here, to run repeatedly:
}`

unCleanCode, 2021-02-24:
Same for me. Also, COL4 doesn't seem to be working
It looks like the problem with all P1.xx pins (P1.00-P1.09)

unCleanCode, 2021-02-24:
Btw, in order to make micro:bit v2 board work I was required to fix boards.txt file:

```
# BBCmicrobitV2.menu.softdevice.none=None
# BBCmicrobitV2.menu.softdevice.none.softdevice=none
# BBCmicrobitV2.menu.softdevice.none.softdeviceversion=
BBCmicrobitV2.menu.softdevice.s132=S132
BBCmicrobitV2.menu.softdevice.s132.softdevice=s132
BBCmicrobitV2.menu.softdevice.s132.softdeviceversion=2.0.1
BBCmicrobitV2.menu.softdevice.s132.upload.maximum_size=409600
BBCmicrobitV2.menu.softdevice.s132.build.extra_flags=-DNRF52 -DS132 -DNRF51_S132
BBCmicrobitV2.menu.softdevice.s132.build.ldscript=armgcc_s132_nrf52832_xxaa.ld
```

unCleanCode, 2021-02-24:
Btw, S132 is already updated to 7.2.0, why still use 2.0.1 ?

unCleanCode, 2021-02-24:
I might be wrong, but I see that nRF52.h struct NRF_GPIO_Type has field:
`PIN_CNF[32];`

which means that when we do:
```
 NRF_GPIO_Type* port = digitalPinToPort(6);
 uint32_t pin = digitalPinToPin(6);
```
we'll get pin == 37

unCleanCode, 2021-02-24:
Ok, I was able to make it work (both the LOGO pin and all P1 pins) by commenting in file:
components/device/nrf.h
line #95:

```
/* Redefine "old" too-generic name NRF52 to NRF52832_XXAA to keep backwards compatibility. */
#if defined (NRF52)
    #ifndef NRF52832_XXAA
        // #define NRF52832_XXAA
    #endif
#endif

```

Polyphe, 2021-02-26:
HI @unCleanCode,
Thank you for your replies, but from my side, it seems not to work. When I comment #define NRF52832_XXAA, with S132 build option, I got:
#error "Device must be defined. See nrf.h."
#error "Unsupported GPIO_COUNT"

Then with build option "nothing" it doesn't work too.
How do you proceed to do working it ?

Thank you in advance.

unCleanCode, 2021-02-26:
Hi @Polyphe 
few notes:

Note #1. The problem for me was that both NRF52832_XXAA and NRF52833_XXAA were defined, and some defines were conflicting (specifically the GPIO_COUNT was 1, when for nRF52833 it should be 2, otherwise it won't see those P1.xx pins)
To check that I did run this simple function:
`void printDef() {
#if defined(NRF51)
  Serial.println("NRF51");
#endif
#if defined(NRF52)
  Serial.println("NRF52");
#endif
#if defined (NRF52805_XXAA)
  Serial.println("NRF52805");
#endif
#if defined(NRF52810_XXAA)
  Serial.println("NRF52810");
#endif
#if defined(NRF52811_XXAA)
  Serial.println("NRF52811");
#endif
#if defined(NRF52820_XXAA)
  Serial.println("NRF52820");
#endif
#if defined(NRF52832_XXAA) 
  Serial.println("NRF52832 XXAA");
#endif
#if defined(NRF52832_XXAB)
  Serial.println("NRF52832 XXAB");
#endif
#if defined (NRF52833_XXAA)
  Serial.println("NRF52833");
#endif
#if defined(NRF52840_XXAA)
  Serial.println("NRF52840");
#endif
#if defined (NRF5340_XXAA_APPLICATION)
  Serial.println("NRF5340");
#endif
#if defined (NRF5340_XXAA_NETWORK)
  Serial.println("NRF5340_NET");
#endif
#if defined(NRF9160_XXAA)
  Serial.println("NRF9160");
#endif
}`

It shown me that NRF52, NRF52832_XXAA and NRF52833_XXAA are defined. 
I found that the NRF52832_XXAA was defined because of that line which I commented and now all works fine
In your case, I suspect, you'll have just NRF52 and NRF52832_XXAA defined (or if you commented this line - just NRF52)

It's pretty odd that in your case you don't have NRF52833_XXAA  defined (please confirm if that's true), as boards.txt file of this project contains this line:
BBCmicrobitV2.build.extra_flags=-DNRF52833_XXAA

Are you using Android IDE or Platformio...or something else?
Whenever you use, please make sure you specified the BBC MicroBit V2 board

Note #2: 
Even after fixing it for myself `attachInterrupt` didn't work with the LOGO pin. But I was able to digitalRead from that pin with success. No idea yet why they attachInterrupt doesn't work with that specific pin, as it works fine with button pins and I also able to read manually from that pin...but, tbh, I have no idea yet how the `attachInterrupt` works internally, maybe it relies on some internal events sent from hardware ...no idea

84828093486894389686896, 2025-07-24:
> Btw, S132 is already updated to 7.2.0, why still use 2.0.1 ?

How did you get softdevice on the micro:bit V2?