sandeepmistry/arduino-nRF5 #457: NRF52840

tcpipchip, 2021-04-11:
Today
Finally i got success to run Sandeepmistry on NRF52840 u-blox module :)
If someone is interested, i can send the package

tcpipchip, 2021-04-11:
Make some upgrades now, to compile bleperipheral with NRF52840
Do you know wich S140 softdevice version will the compatible with NORDIC SDK used by Sandeepmistry ?

tcpipchip, 2021-04-11:
all is working, except some incompatiblity beteween SDK + S140 softdevice version, because BLE not start up

il300, 2021-06-09:
> Today
> Finally i got success to run Sandeepmistry on NRF52840 u-blox module :)
> If someone is interested, i can send the package

i am interested 👍

AccelElec, 2023-04-17:
What softdevice are you using? I can't figure out how to get the s140 softdevice to show in the IDE.

tcpipchip, 2023-04-17:
I  know that is off topic, but why dont you use adafruit ble  on 840 ?

AccelElec, 2023-04-17:
Unfortunately, the Adafruit version doesn't support custom boards nor the ST-Link debugger. I was really hoping to use the ST-Link v4 as it can be soldered directly to a test rig so programming and testing can be done in a single step. Thi will really cut down on production time if I can get this working.

tcpipchip, 2023-04-17:
i had used in the past the
softdevice_s140.hex

AccelElec, 2023-04-17:
Which, the Adafruit version or the ST-Link? On a custom board?

tcpipchip, 2023-04-17:
i remember that worked on sandeep with nrf52840 (u-blox nina b302)