sandeepmistry/arduino-nRF5 #465: How can I create a custom board variant for NRF 52840 and load the Arduino bootloader on this chip?

onderbedirhan, 2021-10-08:
I want to design a custom PCB using the NRF52840 chip. And I want to burn Arduino bootloader on this chip. At this point, when I examined this github repository, I saw that there are many board variants. At this point, what do I need to do step by step to create a custom board variant and upload the Arduino bootloader on this chip? (I would be very happy if you explain in detail). And when I create my custom board variant, can I use all the features of the NRF chip (nfc, i2s etc.) without any problems?

drmpf, 2022-02-25:
have a look at https://www.forward.com.au/pfod/BLE/LowPower/NanoReplacement/index.html#creating
for how to do this for an nRF52383. 
BUT I don't think this repository supports nRF25840 
You can look at https://learn.adafruit.com/adafruit-feather-sense/arduino-support-setup, but it uses a specialized bootloader.

AccelElec, 2023-04-17:
Maybe I'm missing something but this repo includes system_nrf52840.c so doesn't that imply that the nrf52840 is supported? Maybe this is a recent addition?

drmpf, 2023-04-17:
> nrf52840 is supported? Maybe this is a recent addition?
Looks like it. Well atleast after 2018 when I did my low power mods

AccelElec, 2023-04-17:
I've replied to several of these issues with similar questions on the repo; I figured opening a new issue would be redundant. Hopefully, I can get some clarification from someone in the know.