sandeepmistry/arduino-nRF5 #480: nRF52833 to use NRF_P1 problem

mrespin, 2022-03-25:
Hi,
I am using nRF52833, but when the family is set to NRF52, nrf.h is set NRF52832 in line 93 and I cannot use GPIO over #32:
#if defined (NRF52)
    #ifndef NRF52832_XXAA
        #define NRF52832_XXAA  // commenting this line fix
    #endif
#endif
if I uncomment this all working fine.
PS I did not find #define NRF52
I propose to change #ifndef NRF52832_XXAA to check other MCU NRF52 family?
Thank you

carlosperate, 2022-04-11:
Are you using the latest release? It should have something this block instead:
https://github.com/sandeepmistry/arduino-nRF5/blob/75baa76b21ec28cf7c6ece57ba444cf2306b584b/cores/nRF5/SDK/components/device/nrf.h#L92-L97

carlosperate, 2022-04-11:
It looks like variants like the micro:bit V2 simply use the integer value (P1.0 -> 32):
https://github.com/sandeepmistry/arduino-nRF5/blob/75baa76b21ec28cf7c6ece57ba444cf2306b584b/variants/BBCmicrobitV2/variant.cpp#L35

mrespin, 2022-07-18:
Unfortunately, P1.00 is not activated when set to #define NRF52832_XXAA, as NRF52832_XXAA do not have it.
Moreover, I started to use NRF52805, part of the NRF52 family and it lacks FPU and caching mechanism, thus default #define NRF52832_XXAA breaks it even more.
Fix:
#if !defined(NRF52832_XXAA) && !defined(NRF52805_XXAA) && !defined(NRF52833_XXAA) and so on.