sandeepmistry/arduino-nRF5 #481: micro:bit v2 no valid JTAG interface configured error on Macbook Air M1

BereketTesfa, 2022-03-26:
I can not upload Sketchs to the Micro:bit v2 from a Macbook Air M1. It keeps saying the following error messsage:

Error: No Valid JTAG Interface Configured.

Arduino: 1.8.19 (Mac OS X), Board: "BBC micro:bit V2, None"

Sketch uses 2900 bytes (0%) of program storage space. Maximum is 524288 bytes.
Open On-Chip Debugger 0.10.0-dev-gdc53227 (2016-04-09-13:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
swd
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
Error: CMSIS-DAP: SWD not supported
Error: No Valid JTAG Interface Configured.
Error: No Valid JTAG Interface Configured.

carlosperate, 2022-06-16:
Have you got `CMSIS-DAP` selected as the programmer?
I assume in this case OpenOCD must be running under Rosetta 2, as this board doesn't have binaries for the M1 architecture.
Are you able to programme any other boards using OpenOCD?

RichSj, 2022-08-26:
Hello Carlosperate,

I got same problem.
I did CMSIS-DAP selected as the programmer, it is same.

My case use Windows 10, follow : "https://learn.adafruit.com/use-micro-bit-with-arduino?view=all"
Set my Arduino Preferences to "https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json"
The use MicroBit V2 hardware, software set board to BBC microbit V2

[image]

I don't know have use the new code for Arduino, could you create one way like  "https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json".
The I can try my side.

For me case, the early code work for BBC Microbit V1.5 is fine.

Thank you!

Richard

carlosperate, 2022-08-30:
The seetings looks alright. It shouldn't matter, but just in case, make sure you select the right serial por for the micro:bit.

What error message does it print?

Have you changed the micro:bit firmware? There is no need to change, just wanted to make sure it running the default:
https://microbit.org/get-started/user-guide/firmware/

martinwork, 2022-08-30:
It seems to work with micro:bit V2.21 on Mac mini (M1, 2020) with macOS 12.5.1, with these settings. It also worked on Windows 10. I had no success at first with micro:bit V1.5, but it started working after I flashed the micro:bit out-of-box software.

[image]

martinwork, 2022-08-30:
I haven't checked, but the LED blinking code in the default example may not be right for micro:bit V2.

martinwork, 2022-08-30:
Also tested that it works from Windows 10 with micro:bit V2.20.
```
void setup() {  
  Serial.begin(115200);
  
  Serial.println("setup");
  delay(1000);
}

void loop(){
  Serial.println("loop");
  delay(500);
}
```

RichSj, 2022-08-31:
Hello Carlosperate and Martin,

Thank you very much for your support!

I use Windows 10 Arduino for  LED blink code got error:

Arduino: 1.8.19 (Windows 10), Board: "BBC micro:bit V2, None"

Sketch uses 1832 bytes (0%) of program storage space. Maximum is 524288 bytes.

Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
swd
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
Error: CMSIS-DAP: SWD not supported
Error: No Valid JTAG Interface Configured.
Error: No Valid JTAG Interface Configured.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

The difference between my side and Martin is, my side port only support serial port, Martin side got one usbmoden2022 support Microbit V2.

[image]

Could you provide more detail, how include this package in Arduino?
Where I can got this package?

I think this information can help every MicroBit V2 user use Arduino-MicroBit.

Thank you!

Richard

martinwork, 2022-08-31:
I'm no expert, I'm afraid! I followed instructions at https://learn.adafruit.com/use-micro-bit-with-arduino/install-board-and-blink. usbmoden2022 above is because that was from macOS. I am ignoring an update message when the app starts. I'll see what updating does.

Below is Windows 10 Tools menu and output (substituted <user>) after enabling Preferences/Show verbose output during upload. 

[image]

```
Sketch uses 2708 bytes (0%) of program storage space. Maximum is 524288 bytes.
<user>\AppData\Local\Arduino15\packages\sandeepmistry\tools\openocd\0.10.0-dev.nrf5/bin/openocd.exe -d2 -f interface/cmsis-dap.cfg -c transport select swd; -f target/nrf52.cfg -c program {{<user>\AppData\Local\Temp\arduino_build_377683/sketch_aug30a.ino.hex}} verify reset; shutdown; 
Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselect 'swd'
Warn : Transport "swd" was already selected
swd
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 2.1.0
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : reduce speed request: 10000kHz to 5000kHz maximum
Info : clock speed 10000 kHz
Info : SWD IDCODE 0x2ba01477
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
nrf52.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000a80 msp: 0x20000400
** Programming Started **
auto erase enabled
Warn : Unknown device (HWID 0x000001a0)
Warn : using fast async flash loader. This is currently supported
Warn : only with ST-Link and CMSIS-DAP. If you have issues, add
Warn : "set WORKAREASIZE 0" before sourcing nrf51.cfg to disable it
wrote 4096 bytes from file <user>\AppData\Local\Temp\arduino_build_377683/sketch_aug30a.ino.hex in 0.551951s (7.247 KiB/s)
** Programming Finished **
** Verify Started **
verified 2832 bytes in 0.222998s (12.402 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked

```

RichSj, 2022-08-31:
It is weird, in my side, always got :

**Error: CMSIS-DAP: SWD not supported**

These are enabling Preferences/Show verbose output:

"C:\\Users\\Rich_Desktop\\AppData\\Local\\Arduino15\\packages\\builtin\\tools\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Rich_Desktop\\AppData\\Local\\Temp\\arduino_build_327093\\preproc\\ctags_target_for_gcc_minus_e.cpp"

Compiling sketch...

"C:\\Users\\Rich_Desktop\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\tools\\gcc-arm-none-eabi\\5_2-2015q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=64000000 -DARDUINO=10819 -DARDUINO_BBC_MICROBIT_V2 -DARDUINO_ARCH_NRF5 -DNRF52833_XXAA -DNRF5 "-IC:\\Users\\Rich_Desktop\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\cores\\nRF5/SDK/components/toolchain/" "-IC:\\Users\\Rich_Desktop\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\cores\\nRF5/SDK/components/toolchain/CMSIS/Include" "-IC:\\Users\\Rich_Desktop\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\cores\\nRF5/SDK/components/toolchain/gcc/" "-IC:\\Users\\Rich_Desktop\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\cores\\nRF5/SDK/components/device/" "-IC:\\Users\\Rich_Desktop\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\cores\\nRF5/SDK/components/drivers_nrf/delay/" "-IC:\\Users\\Rich_Desktop\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\cores\\nRF5/SDK/components/softdevice/none/headers/" -DUSE_LFSYNT "-IC:\\Users\\Rich_Desktop\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\cores\\nRF5" "-IC:\\Users\\Rich_Desktop\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\variants\\BBCmicrobitV2" "C:\\Users\\Rich_Desktop\\AppData\\Local\\Temp\\arduino_build_327093\\sketch\\Blink.ino.cpp" -o "C:\\Users\\Rich_Desktop\\AppData\\Local\\Temp\\arduino_build_327093\\sketch\\Blink.ino.cpp.o"

Compiling libraries...

Compiling core...

Using previously compiled file: C:\Users\Rich_Desktop\AppData\Local\Temp\arduino_build_327093\core\variant.cpp.o

Using precompiled core: C:\Users\Rich_Desktop\AppData\Local\Temp\arduino_cache_491390\core\core_sandeepmistry_nRF5_BBCmicrobitV2_softdevice_none_d03f356af6979ec169f3015dbcf7deea.a

Linking everything together...

"C:\\Users\\Rich_Desktop\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\tools\\gcc-arm-none-eabi\\5_2-2015q4/bin/arm-none-eabi-gcc" "-LC:\\Users\\Rich_Desktop\\AppData\\Local\\Temp\\arduino_build_327093" -Os -Wl,--gc-sections -save-temps "-LC:\\Users\\Rich_Desktop\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\cores\\nRF5/SDK/components/toolchain/gcc/" "-LC:\\Users\\Rich_Desktop\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\cores\\nRF5/SDK/components/softdevice/none/toolchain/armgcc/" -Tnrf52833_xxaa.ld "-Wl,-Map,C:\\Users\\Rich_Desktop\\AppData\\Local\\Temp\\arduino_build_327093/Blink.ino.map" --specs=nano.specs --specs=nosys.specs -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -o "C:\\Users\\Rich_Desktop\\AppData\\Local\\Temp\\arduino_build_327093/Blink.ino.elf" "C:\\Users\\Rich_Desktop\\AppData\\Local\\Temp\\arduino_build_327093\\sketch\\Blink.ino.cpp.o" "C:\\Users\\Rich_Desktop\\AppData\\Local\\Temp\\arduino_build_327093\\core\\variant.cpp.o" -Wl,--start-group -lm "C:\\Users\\Rich_Desktop\\AppData\\Local\\Temp\\arduino_build_327093/..\\arduino_cache_491390\\core\\core_sandeepmistry_nRF5_BBCmicrobitV2_softdevice_none_d03f356af6979ec169f3015dbcf7deea.a" -Wl,--end-group

"C:\\Users\\Rich_Desktop\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\tools\\gcc-arm-none-eabi\\5_2-2015q4/bin/arm-none-eabi-objcopy" -O binary "C:\\Users\\Rich_Desktop\\AppData\\Local\\Temp\\arduino_build_327093/Blink.ino.elf" "C:\\Users\\Rich_Desktop\\AppData\\Local\\Temp\\arduino_build_327093/Blink.ino.bin"

"C:\\Users\\Rich_Desktop\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\tools\\gcc-arm-none-eabi\\5_2-2015q4/bin/arm-none-eabi-objcopy" -O ihex "C:\\Users\\Rich_Desktop\\AppData\\Local\\Temp\\arduino_build_327093/Blink.ino.elf" "C:\\Users\\Rich_Desktop\\AppData\\Local\\Temp\\arduino_build_327093/Blink.ino.hex"

"C:\\Users\\Rich_Desktop\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\tools\\gcc-arm-none-eabi\\5_2-2015q4/bin/arm-none-eabi-size" -A "C:\\Users\\Rich_Desktop\\AppData\\Local\\Temp\\arduino_build_327093/Blink.ino.elf"

Sketch uses 1832 bytes (0%) of program storage space. Maximum is 524288 bytes.

C:\Users\Rich_Desktop\AppData\Local\Arduino15\packages\sandeepmistry\tools\openocd\0.10.0-dev.nrf5/bin/openocd.exe -d2 -f interface/cmsis-dap.cfg -c transport select swd; -f target/nrf52.cfg -c program {{C:\Users\Rich_Desktop\AppData\Local\Temp\arduino_build_327093/Blink.ino.hex}} verify reset; shutdown; 

Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselect 'swd'
Warn : Transport "swd" was already selected
swd
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
Error: CMSIS-DAP: SWD not supported
Error: No Valid JTAG Interface Configured.
Error: No Valid JTAG Interface Configured.

[image]

Even I tried Arduino LDE V2.  I still got same error.

Thank you!

Richard

carlosperate, 2022-08-31:
Maybe something is not quite right with the micro:bit firmware, could you share here the `DETAILS.TXT` file from the `MICROBIT` drive?

RichSj, 2022-08-31:
Thank you, Carlosperate and Martin!

Sorry I don't know where is the DETAILS.TXT drive located.

There one more detail, it show firmware is good.
Use Microsoft BBC MakeCode, it is good for this board:

[image]

RichSj, 2022-08-31:
Hello Carlosperate and Martin,

I am sure is this you requested?

[Details.zip](https://github.com/sandeepmistry/arduino-nRF5/files/9463959/Details.zip)

carlosperate, 2022-08-31:
I'm not sure what that file is, but when you connect the micro:bit, don't you get a USB drive called `MICROBIT`? Or is it calles something else like `MAINTENANCE`?
Inside that USB drive there should be a file named `DETAILS.TXT`, can you see it?

RichSj, 2022-08-31:
In device list, it already show CMSIS-DAP v2, looks it is good for CMSIS-DAP.

RichSj, 2022-08-31:
[image]

carlosperate, 2022-08-31:
Make sure you uninstall any Mbed drivers you might have manually installed (there was an installer available for older versions of Windows, but in Windows 10 in can interfeer with the other drivers).
There are some instructions at the bottom of this page: https://support.microbit.org/support/solutions/articles/19000105428-webusb-troubleshooting

Did you find the `DETAILS.TXT` file?

RichSj, 2022-08-31:
Sorry, I thought the detail already in picture.
Then I did search these directories (include subdirectories) : Windows; Program Files (X86) ;Program Files, no this file.

I will work for your Mbed suggestion now.

martinwork, 2022-08-31:
Please try excluding the MICROBIT drive from being scanned by your antivirus.

The DETAILS.TXT file is in the MICROBIT drive in Windows Explorer

[image]

RichSj, 2022-08-31:
Thank you, Martin!
Attached:
[DETAILS.TXT](https://github.com/sandeepmistry/arduino-nRF5/files/9464531/DETAILS.TXT)

RichSj, 2022-08-31:
I did disable the antivirus, no help.
I try remove the Mbed driver, but don't know why, I can't touch it even use Administrators account.
Then I use disable this device, no help.

RichSj, 2022-08-31:
This is MicroBit V1.5 Details.txt, it work fine for Arduino,
[V1_DETAILS.TXT](https://github.com/sandeepmistry/arduino-nRF5/files/9464589/V1_DETAILS.TXT)

RichSj, 2022-08-31:
This is Micro Bit V1.5 upload message:

For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselect 'swd'
Warn : Transport "swd" was already selected
swd
cortex_m reset_config sysresetreq
adapter speed: 1000 kHz
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 1.10
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Info : SWD IDCODE 0x0bb11477
Info : nrf51.cpu: hardware has 4 breakpoints, 2 watchpoints
nrf51.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x000019e0 msp: 0x20004000
** Programming Started **
auto erase enabled
Info : nRF51822-QFAA(build code: H0) 256kB Flash
Warn : using fast async flash loader. This is currently supported
Warn : only with ST-Link and CMSIS-DAP. If you have issues, add
Warn : "set WORKAREASIZE 0" before sourcing nrf51.cfg to disable it
wrote 2048 bytes from file C:\Users\Rich_Desktop\AppData\Local\Temp\arduino_build_890142/Blink.ino.hex in 0.307958s (6.494 KiB/s)
** Programming Finished **
** Verify Started **
verified 1508 bytes in 0.091993s (16.008 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked

martinwork, 2022-08-31:
Try this... if you haven't already...

- Run **Export compiled binary** from the **Sketch** menu.
- Look for the file <sketch name>.ino.BBCmicrobitV2.hex next to your <sketch name>.ino file.
- Drag the hex file to the MICROBIT drive to upload it to micro:bit.

RichSj, 2022-08-31:
No error message, but look inside, the file didn't copy in.

martinwork, 2022-08-31:
When you drag a hex file to the MICROBIT drive, it appears there only while it is being processed. The MICROBIT drive will disappear and reappear as the micro:bit resets itself. With this simple sketch, I can see the output in a serial terminal.

[helloserial.zip](https://github.com/sandeepmistry/arduino-nRF5/files/9464854/helloserial.zip)

RichSj, 2022-08-31:
Yes, use your file, I got same, serial port keep output "loop"

martinwork, 2022-09-01:
I would be tempted to reinstall the software, after uninstalling it and making sure that everything related is cleaned out. (e.g.%LocalAppData%\Arduino15, %Temp%), but I know that I won't lose anything by doing that.

RichSj, 2022-09-01:
Hello Martin,

I did, uninstall the Arduino 1.8, delete all file under %LocalAppData%\Arduino15, re-install Arduino, re-add package, the SWD error keep coming.
I did use Arduino for ESP32 , maybe it cause something conflict.

Thank you!

carlosperate, 2022-10-07:
Hi @RichSj, are you still experiencing this problem? Could you post a picture of the back of the micro:bit (where the chips are placed)?

RichSj, 2022-10-11:
Hello Carlos,

I still have problem for the SWD issue when use new V2 Arduino IDE
(Update 3 weeks ago).
But I made one mistake, I thought V2.0 and V2.2 are same, so I notice
it is V2, it is V2.2.
Could you try V2.2 in your side?

Thank you!

Richard Li

carlosperate, 2022-10-11:
V2.2 works for me and for Martin (some of the responses before are from tests using a V2.2). 

Could you try again with your V2.2 board, copy and paste the full output here, together with the details.txt file from the MICROBIT drive and a picture of the back of the board? As there is so much info around here going back and forth it will be really useful to have it all recollected again in a single message.

(Btw, email attachments are not sent to GitHub issues, so you'd have to use the GitHub website to add files in a comment).

RichSj, 2022-10-12:
[MicroBit_22.zip](https://github.com/sandeepmistry/arduino-nRF5/files/9767859/MicroBit_22.zip)
Hello Carlosperate,
Attached the information you requested.

Thank you!

Richard

RichSj, 2022-10-12:
BTW, where I can find V2.2 PCB layout information?
I want know where is the test point location.