sandeepmistry/arduino-nRF5 #482: nrf51822 generic board dose nor advertise Bluetooth at all

AlirezaDadrass, 2022-04-04:
Hi
I am working on a prototype with nrf51822 QFAAC0 and aft 30 days of working and searching for solution I could not manage to run a working program on this chip. 
It works fine as a regular microcontroller but when I try to enable Bluetooth with S110 or S130 (almost all versions that released), Bluetooth did not advertise at all!!
It is weird because I tried mbed os v2 (deprecated and not supported in offline IDEs) it worked, but your library and Bleperipheral Did not.
(I bought nrf52832 module and its fine but in my country it's really hard to access to this chip . I have to design my own PCB and access to available materials is important to me)
If you can help with I will be appreciated

TiNredmc, 2022-09-03:
I also having similar problem. I bought WT51822-S4AT module back in 2021. The chip is QFAAG1 which is working just fine (including BLE). Last month I bought 2 WT51822-S4AT module again. But this time the chip is QFAAH0. And despite marking build code as H0, openOCD report as either CA or C0.
`Info : nRF51822-QFAA(build code: CA/C0) 256kB Flash`
same problem as your. BLE is not functioning.

TiNredmc, 2022-09-06:
Update : I did some more digging. For some reason. Both of my WT51822-S4AT has QFAAC0 and hardware ID 0x001d, which is revision 1. Still have no Idea why the package is labeled H0. That might explain why BLE won't work due to SDK and soft device compatibility.

Houba28, 2023-05-17:
@TiNredmc So is it working for you? I have also WT51822-S4AT, dont know the chip precisely now, but it seems that the program crashes when executing the line with BLESerial.begin(). To find this out, I just connected two leds, and lit one before this line, and second after this line. The second did not lit up. I am looking for any help or advice, how to find out what is happening since I dont know any way to debug. Using st-link V2 as a programmer. Tried both this library, and also the suggested BLEPheripheral with the same results. Thanks in advance for any help.

TiNredmc, 2023-05-17:
@Houba28 Hi. The problem made me ditch the nRF51 on the arduino entirely. I planned to use the official SDK instead (not any time soon). But now I'm using the WCH CH582F as an alternative. I have to rebuild/recreate the project from the Arduino nRF51.

TiNredmc, 2023-12-16:
@Houba28 
Several days ago. I came back to the nRF51822 after my friend got 2 of these to use in their project. It reminds me of my own WT51822-S4AT modules (That I bought the H0 chip to replace the C0). I got mine working with S110 right away. But I have to adjust the IRAM reservation for S130 to make it work. 

I edited the linker script that came with the S130 Soft device that buried deep into the Arduino package folder.
The RAM **ORIGIN** has to increase to 10K. Which means that soft device will reserve the RAM for 10K, Left 6K for the Application code
```
MEMORY
{
  FLASH (rx) : ORIGIN = 0x1b000, LENGTH = 0x25000
  RAM (rwx) :  ORIGIN = 0x20000000 + 10K, LENGTH = 16K - 10K
}
``` 

Not so sure that this will work with C0 chip (as I no longer have a C0 board to test with anymore).