sandeepmistry/arduino-nRF5 #497: Error while flashing SoftDevice... Server returned HTTP response code: 403

Teddyz, 2022-11-01:
I have a board with nRF51822 (CJMCU WCMCU-8223), bought a few years back and a ST-LINK V2 adapter from ebay. 

When I try flashing either S110 or S130 I get one of these errors:
```
Downloading 'https://www.nordicsemi.com/-/media/Software-and-other-downloads/SoftDevices/S110/s110nrf51800.zip' ...
Error while flashing SoftDevice.
java.io.IOException: Server returned HTTP response code: 403 for URL: https://www.nordicsemi.com/-/media/Software-and-other-downloads/SoftDevices/S110/s110nrf51800.zip
Downloading 'https://www.nordicsemi.com/-/media/Software-and-other-downloads/SoftDevices/S130/s130nrf51201.zip' ...
Error while flashing SoftDevice.
java.io.IOException: Server returned HTTP response code: 403 for URL: https://www.nordicsemi.com/-/media/Software-and-other-downloads/SoftDevices/S130/s130nrf51201.zip

```
Is it just me, or are the files down for everyone?

Other questions I have are:
  Are there any documentation on my board?
  Would it be much helpful for me if my board was included among the predefined in arduino-nRF5?
  Do I need drivers on Win7 for the ST-LINK thing?

carlosperate, 2022-11-07:
Both URLs work for me today, it was maybe a temporary issue in the Nordic Semi servers.

Even if the SoftDevice flashing didn't work at the time, were you able to flash a simple blinkly programme with the ST-Link?

Teddyz, 2022-11-10:
Thank you!
Now the links works for me too. 

I was able to flash the device from Arduino IDE. Now I have working BLE as well. 
Next step is to understand what I am doing. :)

closed 2022-11-10 by Teddyz

KishanKartha, 2023-12-09:
> Thank you! Now the links works for me too.
> 
> I was able to flash the device from Arduino IDE. Now I have working BLE as well. Next step is to understand what I am doing. :)

Hi Teddyz.

Were you able to successfully work with the CJMCU-8223 board? I am currently working on it. It is really confusion where to start.