sandeepmistry/arduino-nRF5 #501: stdint.h Missing

bbburnaby, 2023-01-08:
I am trying to code my microbit on Arduino IDE (for the first time). I keep getting the same message whenever I try to upload one of the example files (e.g. "Microbit Compass demo" or "blinkdemo"). Can anyone help me out with this?

Arduino: 1.8.19 (Windows 10), TD: 1.56, Board: "BBC micro:bit, S110"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\*\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\*\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\*\Documents\Arduino\libraries -fqbn=sandeepmistry:nRF5:BBCmicrobit:softdevice=s110 -vid-pid=0D28_0204 -ide-version=10819 -build-path C:\Users\*\AppData\Local\Temp\arduino_build_168042 -warnings=none -build-cache C:\Users\*\AppData\Local\Temp\arduino_cache_789952 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.gcc-arm-none-eabi.path=C:\Users\*\AppData\Local\Arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4 -prefs=runtime.tools.gcc-arm-none-eabi-5_2-2015q4.path=C:\Users\*\AppData\Local\Arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4 -prefs=runtime.tools.openocd.path=C:\Users\*\AppData\Local\Arduino15\packages\sandeepmistry\tools\openocd\0.10.0-dev.nrf5 -prefs=runtime.tools.openocd-0.10.0-dev.nrf5.path=C:\Users\*\AppData\Local\Arduino15\packages\sandeepmistry\tools\openocd\0.10.0-dev.nrf5 -verbose C:\Users\*\Documents\Arduino\libraries\MicrobitV2-HHS\examples\MicrobitV2_Acc_Compass\MicrobitV2_Acc_Compass.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\*\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\*\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\*\Documents\Arduino\libraries -fqbn=sandeepmistry:nRF5:BBCmicrobit:softdevice=s110 -vid-pid=0D28_0204 -ide-version=10819 -build-path C:\Users\*\AppData\Local\Temp\arduino_build_168042 -warnings=none -build-cache C:\Users\*\AppData\Local\Temp\arduino_cache_789952 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.gcc-arm-none-eabi.path=C:\Users\*\AppData\Local\Arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4 -prefs=runtime.tools.gcc-arm-none-eabi-5_2-2015q4.path=C:\Users\*\AppData\Local\Arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4 -prefs=runtime.tools.openocd.path=C:\Users\*\AppData\Local\Arduino15\packages\sandeepmistry\tools\openocd\0.10.0-dev.nrf5 -prefs=runtime.tools.openocd-0.10.0-dev.nrf5.path=C:\Users\*\AppData\Local\Arduino15\packages\sandeepmistry\tools\openocd\0.10.0-dev.nrf5 -verbose C:\Users\*\Documents\Arduino\libraries\MicrobitV2-HHS\examples\MicrobitV2_Acc_Compass\MicrobitV2_Acc_Compass.ino

Using board 'BBCmicrobit' from platform in folder: C:\Users\*\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.7.0

Using core 'nRF5' from platform in folder: C:\Users\*\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.7.0

Detecting libraries used...

"C:\\Users\\*\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\tools\\gcc-arm-none-eabi\\5_2-2015q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=16000000 -DARDUINO=10819 -DARDUINO_BBC_MICROBIT -DARDUINO_ARCH_NRF5 -DNRF51 -DS110 -DNRF51_S110 -DNRF5 "-IC:\\Users\\*\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\cores\\nRF5/SDK/components/toolchain/" "-IC:\\Users\\*\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\cores\\nRF5/SDK/components/toolchain/CMSIS/Include" "-IC:\\Users\\*\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\cores\\nRF5/SDK/components/toolchain/gcc/" "-IC:\\Users\\*\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\cores\\nRF5/SDK/components/device/" "-IC:\\Users\\*\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\cores\\nRF5/SDK/components/drivers_nrf/delay/" "-IC:\\Users\\*\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\cores\\nRF5/SDK/components/softdevice/s110/headers/" -DUSE_LFRC "-IC:\\Users\\*\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\cores\\nRF5" "-IC:\\Users\\*\\AppData\\Local\\Arduino15\\packages\\sandeepmistry\\hardware\\nRF5\\0.7.0\\variants\\BBCmicrobit" "C:\\Users\\*\\AppData\\Local\\Temp\\arduino_build_168042\\sketch\\MicrobitV2_Acc_Compass.ino.cpp" -o nul

Alternatives for stdint.h: []

ResolveLibrary(stdint.h)

  -> candidates: []

In file included from C:\Users\*\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.7.0\cores\nRF5/Arduino.h:5:0,

                 from C:\Users\*\AppData\Local\Temp\arduino_build_168042\sketch\MicrobitV2_Acc_Compass.ino.cpp:1:

c:\users\*\appdata\local\arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4\lib\gcc\arm-none-eabi\5.2.1\include\stdint.h:9:26: fatal error: stdint.h: No such file or directory

compilation terminated.

exit status 1

Error compiling for board BBC micro:bit.

carlosperate, 2023-01-13:
where is the example from? Could you link to the source code?

The error it's a bit odd, as it should be included with the compiler. Have you tried maybe uninstalling the Arduino Core and installing it again? In case something got corrupted in the tools at some point?

bbburnaby, 2023-01-14:
`const int COL1 = 3;     // Column #1 control
const int LED = 26;     // 'row 1' led

void setup() {  
  Serial.begin(9600);
  
  Serial.println("microbit is ready!");

  // because the LEDs are multiplexed, we must ground the opposite side of the LED
  pinMode(COL1, OUTPUT);
  digitalWrite(COL1, LOW); 
   
  pinMode(LED, OUTPUT);   
}

void loop(){
  Serial.println("blink!");
  
  digitalWrite(LED, HIGH);
  delay(500);
  digitalWrite(LED, LOW);
  delay(500);
}`
I have not tried uninstalling and reinstalling the Arduino Core. What is the Arduino Core?