sandeepmistry/arduino-nRF5 #503: Connecting NRF52840 chip

bbuster2, 2023-02-11:
I have bought an NRF52840 module (https://www.ebyte.com/en/product-view-news.aspx?id=445) and i would like to program it via Arduino IDE or VSCode Platformio.
Maybe i am a bit foolish but i see the board (EByte nrf52840) has D+, D-, VBS and GND connections.

Will i be able to just use a USB cable and connect the 4 wires to these pins? And upload code via VSCode or Arduino to the chip?
I see here the USB connections need some resistors and a capacitor and i can add these but maybe i need more than just this:

[image]

Also i read somewhere i need to burn the bootloader and have to use J-Link. That is right or can i do that any other way without J-Link? (Don't have this module) but i do have the ESP-Prog for example

jpconstantineau, 2023-02-16:
I have used this module as the basis of my BlueMicro840 modules.  I use the Adafruit BSP and the PCA10056 (the nRF52840 DK) bootloader and configuration to be able to program it with Arduino.  Unfortunately, to be able to flash the bootloader, you will need either a Jlink or a BlackMagic Probe.  Some people have been able to use OpenOCD to program it with all sorts of SWD programmers but I never was really able to be successful with these unsupported programmers.  

As for your schematic above, you will need to get the SWDIO and SWDCLK pins out and made available to the programmer. (These are the 2 lines at the bottom that are already labelled)  I usually bring them out to a simple header to get a firm/solid connection.

Do bring out the reset pin as well (connected to a button switch to GND).  This will enable you to put the board in bootloader mode.  Have a look at the schematic on this page: https://nrf52.jpconstantineau.com/docs/bluemicro840_v1  This will help you get all the necessary connections and even give you guidance on how to flash the bootloader.

Have a look at the BlueMicro youtube channel and there are examples on how to program these modules.  You will likely need to unlock the module before the bootloader can be flashed.  With both jlink and blackmagic probe, the unlock commands are relatively simple.

bbuster2, 2023-02-17:
Thank you for the detailed information. I am just wondering if the $900 JLink device is what everyone has. I do see the Blackmagic probe but i don’t see where i can buy that and how expensive it is. 

Aliexpress offers “fake” jlink devices but i have no idea if they can work. 

Do you have any information on that?

jpconstantineau, 2023-02-17:
There is a Jink EDU that's inexpensive and is genuine.  Beats going for those Chinese fakes...  The fakes do seem to work  but they do generate odd errors on occasions.  The BMP is designed by 1BitSquared an is on his store.  Adafruit does have them on occasion. With the silicon shortage, these programmers have been affected too (both price and availability).

bbuster2, 2023-02-21:
Thank you, that is more doable but i also saw a way to flash the nRF52840 with an ESP32 (which i have laying around).
(https://github.com/atc1441/ESP32_nRF52_SWD)

I think i can flash the bootloader but the only problem is i cannot get a connection with the nRF52840.
I tried several bootloaders (seeed USB bootloader, Adafruit Feather, etc) and in some cases i see a new USB device being connected, but i cannot upload my code.

I get this:

>Sketch uses 138160 bytes (16%) of program storage space. Maximum is 815104 bytes.
Global variables use 13836 bytes (5%) of dynamic memory, leaving 223732 bytes for local variables. Maximum is 237568 bytes.
Performing 1200-bps touch reset on serial port /dev/cu.usbmodemF2BD185AD80C1
Waiting for upload port...
No upload port found, using address:"/dev/cu.usbmodemF2BD185AD80C1" label:"/dev/cu.usbmodemF2BD185AD80C1" protocol:"serial" protocol_label:"Serial Port (USB)" properties:{key:"pid" value:"0x521F"} properties:{key:"serialNumber" value:"F2BD185AD80C"} properties:{key:"vid" value:"0x1915"} as fallback

Also the /dev/cu.usbmodemF2BD185AD80C1 is gone when i unplug my ESP32 (even with the nRF52840 usb cable connected).

Do you have any suggestion?

jpconstantineau, 2023-02-23:
If you use the [Adafruit_nRF52_Bootloader](https://github.com/jpconstantineau/Adafruit_nRF52_Bootloader), it generally goes into bootloader mode when there are no firmware uploaded to it.  If there are no "USB Drive" showing up, you will need to "double reset" or press the reset "button" twice in a row to get it in bootloader mode.  If that doesn't work, I suspect there is something wrong with your USB connection.

Were you able to flash the bootloader?  If not, remember to unlock the nrf52840 before uploading the bootloader.  I don't know if they still do this but I know in the past, ebyte was locking the code programmed and you couldn't upload anything to it through swd unless the unlock command was sent.  With the Jlink, it's a simple set of commands to flash it:

```
nrfjprog --family NRF52 --recover
nrfjprog --family NRF52 --eraseall
nrfjprog --family NRF52 --program %bootloaderprefix%\%bootloader%  --chiperase --reset
```

With the blackmagic probe, the command is a bit different:
```
arm-none-eabi-gdb.exe -quiet --batch -ex "target extended-remote \\.\COM4" -ex "mon tpwr enable" -ex "mon swdp_scan" -ex "att 1" -ex "mon erase_mass" -ex "mon hard_srst" -ex detach -ex "echo \nUnlock+Erase complete!\n" -ex detach -ex quit 
```
followed by
```
arm-none-eabi-gdb.exe bluemicro_nrf52840_bootloader-0.4.1_s140_6.1.1.hex -quiet --batch -ex "target extended-remote \\.\COM4" -ex "mon tpwr enable" -ex "mon swdp_scan" -ex "att 1" ex "echo \nFlashing\n" -ex load   -ex "echo \nVerifying\n" -ex "compare-sections" -ex "mon hard_srst" -ex detach -ex "echo \nBootloader upload complete!\n" -ex quit 
```

With other programmers, the commands used will likely be different - and as such, I can't help.

bbuster2, 2023-02-25:
Thank you,
I will probably go for the real JLINK, and i see 2 variants with a small price difference. Is there any other difference except the housing?
And will both devices work?

- https://www.antratek.nl/j-link-edu-mini?gclid=Cj0KCQiAgOefBhDgARIsAMhqXA7G71Cy0zwt1McncwjojNS8sj3gYWh5SLjAQgrZ0iyTfnXi5ROQHxgaAouCEALw_wcB
- https://www.sossolutions.nl/1369-segger-j-link-edu-jtagswd-debugger?gclid=Cj0KCQiAgOefBhDgARIsAMhqXA7qbZRWhFNuZxZ1mA44aDxyrT_fYQIXakSNCs_WaoVdfFcirYTSkg8aAt0VEALw_wcB

bbuster2, 2023-02-25:
Update:

I got a bit further. After selecting another Arduino bootloader file (pca10056_bootloader-0.3.0_s140_6.1.1.hex), a new USB device got detected (NRF52BOOT).

I could now upload an example sketch via Arduino IDE and i got this:

```
Opened serial port /dev/cu.usbmodem1301
Starting DFU upgrade of type 4, SoftDevice size: 0, bootloader size: 0, application size: 113640
Sending DFU start packet
Sending DFU init packet
Sending firmware file
########################################
########################################
########################################
########################################
########################################
######################
Activating new firmware

DFU upgrade took 7.148498058319092s
Device programmed.
```

After that, nothing happens (except Mac says the USB device is disconnected).
After double resetting the device, the NRF52BOOT shows up again and i see these 3 files:

- CURRENT.UF2
- INDEX.HTM
- INFO_UF2.TXT

But the serial monitor doesn't show any data, and the nrf beacon app (for ios) says the beacon is not found.
What am i doing wrong?
And just powering the nrf up isn't enough to make it start the normal way? I need to double reset it to get the NRF52BOOT usb device come up

bbuster2, 2023-02-26:
Sorry for my spamming but i hope this is my last question.
I got it working for some time. I could double reset my nRF chip and i got the folder mounting (even in Mac) and when i uploaded my code, i could see the output via the Serial monitor.

I just don't know how i got that working because when i reset the nRF now it supposed to just run (not entering the "download code" mode?) and when pressing the reset button twice i should be able to upload code.

The problem is when i reset the device twice i can see the USB connection on Windows (COMX) or mac ( /dev/cu.usbmodemXXXXXXX) but when i reset it once (to run the code), no USB connection appears so i cannot "debug" my code or see what the Serial monitor outputs.

I tried the Adafruit beacon example but it cannot see any bluetooth device so i am not sure my code "runs".
Am i doing something wrong here?

Thanks again for your patience and support

bbuster2, 2023-03-07:
No idea?

jpconstantineau, 2023-03-08:
The firmware you upload will dictate how it interfaces with usb. If there isn't any code to enable USB functionality it won't have any. Double reset will get the board in bootloader mode and anything you do in firmware won't have any impact.  To test firmware, have you tried a simple led blinking code?