sandeepmistry/arduino-nRF5 #508: Simple blink works but BLE not

beckmx, 2023-05-08:
Hello guys, I installed everything in place, at first I was in doubt that I did it correctly, because none of the examples of the BLE Peripheral work for me, so I tested a simple blink file and the blink does work, then if I want to flash iBeacon or anything related to BLE, looks like it doesnt work.
These are the steps I made so far:

- Installed this core
- Flash the S110, for that I am getting this log:
```
Open On-Chip Debugger 0.10.0-dev-gdc53227 (2016-04-09-13:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
0x4000
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v17 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.250268
Info : nrf51.cpu: hardware has 4 breakpoints, 2 watchpoints
nrf51.cpu: target state: halted
target halted due to debug-request, current mode: Handler SVCall
xPSR: 0x6100000b pc: 0x000024c8 msp: 0x20003cc8
Info : nRF51822-QFAA(build code: H0) 256kB Flash
nrf51.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
auto erase enabled
Info : Padding image section 0 with 2112 bytes
Warn : using fast async flash loader. This is currently supported
Warn : only with ST-Link and CMSIS-DAP. If you have issues, add
Warn : "set WORKAREASIZE 0" before sourcing nrf51.cfg to disable it
nrf51.cpu: target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000001e msp: 0xfffffffc
wrote 93184 bytes from file /Users/xxxx/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.7.0/cores/nRF5/SDK/components/softdevice/s110/hex/s110_nrf51_8.0.0_softdevice.hex in 4.349101s (20.924 KiB/s)
** Programming Finished **
** Verify Started **
nrf51.cpu: target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0xfffffffc
nrf51.cpu: target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0xfffffffc
verified 90328 bytes in 0.587745s (150.084 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```
- Trying to flash the "serial" example and I get this:
```
Open On-Chip Debugger 0.10.0-dev-gdc53227 (2016-04-09-13:45)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
0x4000
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v17 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.250268
Info : nrf51.cpu: hardware has 4 breakpoints, 2 watchpoints
nrf51.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x000006d0 msp: 0x000007c0
** Programming Started **
auto erase enabled
Info : nRF51822-QFAA(build code: H0) 256kB Flash
Warn : using fast async flash loader. This is currently supported
Warn : only with ST-Link and CMSIS-DAP. If you have issues, add
Warn : "set WORKAREASIZE 0" before sourcing nrf51.cfg to disable it
nrf51.cpu: target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000001e msp: 0x000007c0
wrote 32768 bytes from file /var/folders/s5/wx1jrbjd5hbbdwbspxk992n00000gp/T/arduino_build_933570/serial.ino.hex in 1.556717s (20.556 KiB/s)
** Programming Finished **
** Verify Started **
nrf51.cpu: target state: halted
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000002e msp: 0x000007c0
verified 32044 bytes in 0.212495s (147.264 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```

- I restarted the module, but nothing happens. If you can guide me, I would be greateful, thanx!

Mboudali, 2023-08-14:
Not sure of you are still working on this.
What do you mean by nothing happens?

You won't get serial on your computer through the ST-link V2.
Use the nRF Cconnect app and scan for your device. You can use nRF Toolbox -> UART to communicate with the device.

beckmx, 2023-09-06:
I managed to get it working

closed 2023-09-06 by beckmx