sandeepmistry/arduino-nRF5 #510: Only serial works - micro:bit v2.21

WallysFerreira, 2023-07-03:
Using the numbers from the pinmap on the micro:bit website. digitalRead on the buttons pins seems random, digitalWrite to led matrix does not work. Serial.println works though. Any help is appreciated

carlosperate, 2023-07-06:
How are you trying to control the LED matrix? The pin map from the micro:bit website only covers the pins broken out to the edge connector (the columns), but the matrix row signals are not wired out.

[image]

The button are pins 5 and 11 and have external pull ups, so they should read always high, do you have example code to replicate this issue?  

Btw, you can have a look at the schematic here:
https://github.com/microbit-foundation/microbit-v2-hardware/blob/main/V2.21/MicroBit_V2.2.1_nRF52820%20schematic.PDF

And the pin definitions of this Arduino Core are here:
https://github.com/sandeepmistry/arduino-nRF5/blob/master/variants/BBCmicrobitV2/variant.cpp

If you spot any errors a PR would be welcome! :)

metehoca, 2024-02-27:
There's a problem on variant.cpp. Using micro:bit v.2.2.1, trying to blink center LED (COL3-ROW3).

variant.cpp says COL3 = 31 or A3, ROW3 = 15 but nothing happens.

I figured out that COL3 = 3 and ROW3 = 23 works.

carlosperate, 2024-02-27:
I think variant.cpp is probably correct, but maybe you are mixing the Arduino pin numbering with the MCU pin numbering.

The Arduino pins for the rows are indeed: 21, 22, 23, 24, 25
Which in variant.cpp are mapped to the nRF52833 pins: 21, 22, 15, 24, 19
https://github.com/sandeepmistry/arduino-nRF5/blob/8d91cbb022af4f80de2f2131db275e08c9255321/variants/BBCmicrobitV2/variant.cpp#L37-L42

From the schematic:
<p>[image]</p><p>[image]</p><p>[image]</p><p>[image]</p>

So in this case, ROW3 is pin 15 (P0_15) on the nRF52833, which in the Arduino pins is mapped to pin 23.

metehoca, 2024-02-28:
Thanks, it's not obvious at the first sight.