sandeepmistry/arduino-nRF5 #512: Missing 'NRF5 Flash Softdevice' entry in tools menu

Flavelius, 2023-07-27:
Hi,
i have this strange issue where in my arduino ide (2.1.1) there's no 'NRF5 Flash Softdevice' menu entry under the tools menu:
[image]

I've followed the readme and downloaded the jar, placed it in the path: ~/Arduino/tools/nRF5FlashSoftDevice/tool/
and installed the board from the boardmanager. It seems i can almost successfully upload sketches (uploading the blink sketch succeeds, even though nothing blinks on my board afterwards, but that's probably just an LED mapping issue).
Only the softdevice upload tool menu entry is missing.
What could be the issue?

Mboudali, 2023-07-27:
Try IDE 1.8.x

Flavelius, 2023-07-28:
Thanks, that does indeed work. The readme should be updated to mention that this requires the legacy arduino IDE.

Mboudali, 2023-07-31:
What board/nRF are you using? 
Did you manage to pair it with android or windows?

Flavelius, 2023-07-31:
Generic nrf51822, it's a cheap one (holyiot) from eBay just for experimentation. By default it didn't seem to have Bluetooth enabled firmware, but using the legacy ide I was able to flash the example counter which does indeed work as expected and I can subscribe to its updates via the nrf connect app on android

Mboudali, 2023-08-01:
When you pair it with windows, does it say "connected"? Does it create a port? (device manager -> Ports (COM & LPT))
I can't get my nRF52833 to connect. However, I can access it using the nRF connect app.

Flavelius, 2023-08-01:
i'm not on windows, but ubuntu. I'm connecting to it via ST-link V2, which doesn't create a serial port, it connects as just a regular USB device.

Mboudali, 2023-08-04:
Once you finish programming the module and deploy your code. Can you connect to it using Bluetooth?

Flavelius, 2023-08-04:
As i already mentioned, yes, it works.

tomfleet, 2023-10-05:
I am not very sure that this is how BLE serial works.

"Serial" as you are thinking is a butchered adaptation of the GATT
profiles. With data being exchanged via advertising packets.

This takes some glue on the other end, I've never - to my memory - seen
anyone using it to talk to windows like this.

Actuall Bluetooth serial was a thing in older pre-LE profiles. But you'll
probably want to look at the various chrome examples from adafruit etc for
"Web Bluetooth" if you want an easy path to doing what I think you're
trying to do.

You won't get much from the standard windows Bluetooth API here for serial
port usage.

It's not an issue with the Nordic side of things. But an issue with the PC
OS

rogerclarkmelbourne, 2023-10-17:
Did you fix the original problem for this issue.

Did you copy the flashsoftdevice jar into the new folder you need to make according to the instructions in the ReadMe.

Doing this will give you the Flash softdevice menu option, but at the moment it doesn't seem to work, as when it tries to download the softdevice zip file, there is a HTTP error 403

So you have to manually download the S130 or S110 softdevice zip file from Nordic and unzip it and put it onto the appropriate location in the %appdata% local arduino15 core etc etc etc location

Flavelius, 2023-10-19:
if i remember correctly then i did follow all of the logical steps, like moving the manually downloaded file to the required and other folders that seemed logical. But only downgrading the ide helped.