sandeepmistry/arduino-nRF5 #513: Successful upload but non-functional code

samlbakr, 2023-09-16:
I have a custom nRF52810 based board connected via ST-Link v2 programmer. I have attempted using boards _Generic nRF52_ and _nRF52 DK_. Theoretically the nRF52810 should be identical to the nRF52823 on the DK (Besides the memory.) When I flash my code it uploads and verifies successfully however there is no output on my board (Basic blink sketch.) I am quite sure I am using the correct GPIOs and even wrote a test that looped through all the digital pins to be sure. Is the nRF52810 not supported?

Mboudali, 2023-09-26:
Hi Samlbakr, 
Can you share more info please? Wiring, softdevice version, arduino code.
Do you see your device in the "nRF Connect" app?

You may need to look at the documentation of 52810 and make sure the pin mapping matches the Generic nRF52. I had to write my own for 52833.
Page 389: https://infocenter.nordicsemi.com/pdf/nRF52810_PS_v1.4.pdf

samlbakr, 2023-09-26:
> Hi Samlbakr, Can you share more info please? Wiring, softdevice version, arduino code. Do you see your device in the "nRF Connect" app?

Hello, thank you for the reply. My PCB closely follows the QFN32 LDO schematic in the NRF52810 datasheet (pg 400.) with an LED on P0.20. It's powered with a 3.7v li-ion with a 3.3v regulator. I broke out the reset and SWD pins and connected GND, 3.3V, SWDIO, and SWDCLK to the ST-Link v2. I have been flashing with soft device S132 and Generic nRF52 selected. Every time it is has uploaded and verified bytes successfully. I also just bought a j-link to test it out with nRF Connect. (It refused to work with ST Link) It shows up and I can read and write successfully. Tested the j-link with the Arduino IDE and its the same result; upload succeeds but nothing happens.

Here is the code I have been trying to upload:

```
#define LED 15

void setup() 
{
  pinMode(LED, OUTPUT);
}

void loop() 
{
  digitalWrite(LED, HIGH);
}
```

I am not 100% on the pin mapping and have even tried with every integer 1-32 with no result lol. I have taken a peak at the variant files but honestly have no idea what I am looking at. Could you point me towards some resources so that I can write my own variant?

samlbakr, 2023-09-27:
UPDATE: I have successfully gotten a simple blink sketch to work using Segger studio. Still no luck with Arduino. I tried making my own variant but don't really know what I am doing. I really don't want to have to use Segger studio if possible. Any help would be greatly appreciated!

Mboudali, 2023-09-28:
You have thee files in variants: pins_arduino.h, variant.cpp, and variant.h.
You need to modify variant.h and variant.cpp
Start with files from a variant that uses a soc close to the nRF52810, check the datasheet of your module and modify the pin allocation accordingly.

in the soft device folder (ex: C:\Users\MyUserName\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.7.0\cores\nRF5\SDK\components\softdevice), make sure you have a compatible softdevice with your module.

in the text file "boards.txt", add your variant and the appropriate functions. You will find the file in: 
C:\Users\MyUserName\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.7.0

Example:
```
# nRF52833 variants
###################

Generic_nRF52833.name=Generic nRF52833

Generic_nRF52833.upload.tool=sandeepmistry:openocd
Generic_nRF52833.upload.target=nrf52
Generic_nRF52833.upload.maximum_size=524288

Generic_nRF52833.bootloader.tool=sandeepmistry:openocd

Generic_nRF52833.build.mcu=cortex-m4
Generic_nRF52833.build.f_cpu=64000000
Generic_nRF52833.build.board=GENERIC
Generic_nRF52833.build.core=nRF5
Generic_nRF52833.build.variant=Generic
Generic_nRF52833.build.variant_system_lib=
Generic_nRF52833.build.extra_flags=-DNRF52833_XXAA
Generic_nRF52833.build.float_flags=-mfloat-abi=hard -mfpu=fpv4-sp-d16
Generic_nRF52833.build.ldscript=nrf52833_xxaa.ld

Generic_nRF52833.menu.softdevice.none=None
Generic_nRF52833.menu.softdevice.none.softdevice=none
Generic_nRF52833.menu.softdevice.none.softdeviceversion=

Generic_nRF52833.menu.softdevice.s113=S113
Generic_nRF52833.menu.softdevice.s113.softdevice=s113
Generic_nRF52833.menu.softdevice.s113.softdeviceversion=7.2.0
Generic_nRF52833.menu.softdevice.s113.upload.maximum_size=409600
Generic_nRF52833.menu.softdevice.s113.build.extra_flags=-DNRF52 -DS113
Generic_nRF52833.menu.softdevice.s113.build.ldscript=armgcc_s113_nrf52833_xxaa.ld
```

samlbakr, 2023-09-29:
Thank you!! I was able to write a variant for the nRF52810 and my blink sketch now works perfectly. However, I am having some trouble with the softdevice. When I select a softdevice rather than `None`, I get the following error: 

`cannot open linker script file armgcc_s113_nrf52810_xxaa.ld: No such file or directory` 

I was able to find a linker script for s132 which compiles, but I only need peripheral BLE functionality so I was hoping to use the s113. Any clue where I could find this?

Thanks so much again!

Mboudali, 2023-10-11:
Create a folder in 
C:\Users\MyUserName\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.7.0\cores\nRF5\SDK\components\softdevice
and name it "s113".
The folder should have the following structure:
s113
   headers
      nrf52
   hex
   toolchain
       armgcc

In "headers" folder, place the .h files from the softdevice file you downloaded from nRF website. 
In "hex" folder, place the file named "s113_nrf52_X.X.X_softdevice.hex" where X.X.X is the version number.
In "armgcc" folder, create a text file and rename it "armgcc_s113_nrf52810_xxaa.ld" 
Copy/paste the following in the text file:

```
/* Linker script to configure memory regions. */

SEARCH_DIR(.)
GROUP(-lgcc -lc -lnosys)

MEMORY
{
  FLASH (rx) : ORIGIN = 0x1c000, LENGTH = 0x64000
  RAM (rwx) :  ORIGIN = 0x20002080, LENGTH = 0xdf80
}

SECTIONS
{
  .fs_data :
  {
    PROVIDE(__start_fs_data = .);
    KEEP(*(.fs_data))
    PROVIDE(__stop_fs_data = .);
  } > RAM
} INSERT AFTER .data;

INCLUDE "nrf5x_common.ld"
```

Make sure the file names and the properties in "boards.txt" match.

Georg-code, 2024-05-01:
I have the same problem on a custom nrf52820 board. How do I fix that issue? Everything seems like it should work but the soc just does nothing..
Can I use this linker script on that board? How can I create a Linker script fot the **20 board?
Thank you very much

Mboudali, 2024-05-06:
Hi George, 
Did you follow instructions to install nRF52 core/libraries? Did you get any errors during the process?

Could you describe your system?
- OS; Arduino IDE; programmer; nRF soft device 
- Errors when you upload a sketch
- Anything else you think is relevant

positron96, 2024-06-17:
@samlbakr Could you share more info on how you made blink work on nRF52810? I am currently doing the same by writing a custom board.json with necessary options, and I stumble upon 2 issues:
1) Python builder script `nrf5.py` adds `-DNRF52` switch, which is translated into `NRF52832` later in headers. This seems to work ok if I also define `-DNRF52810`, but there are 2 MCU macros defined, which can be a source of problems.
2) `nrf52.h` unconditionally defines __FPU_PRESENT=1 to indicate that an MCU has an FPU, which nRF52810 lacks.
3) a further potential issue (I haven't gotten to that yet) is that python scripts configures GCC with `-mfpu=fpv4-sp-d16`, which may or may not be a problem