sandeepmistry/arduino-nRF5 #515: Flash softdevice jar does not work - HTTP error 403

rogerclarkmelbourne, 2023-10-17:
Not sure why but Nordics site is seems to return HTTP 403 for the softdevice download.

Cutting and pasting the URL shown by the flashsoftdevice jar downloads the zip

The zip can then be opened and the softdevice hex file copied into the correct location in the cores structure

Possibly the UserAgent needs to be set when using Java to download the zip, or if that won't work,  just give instructions about where to download the softdevice(s) and cop them into the cores structure

Its only the download that fails, so once the hex files have been put in the correct location, the Jar doesn't try to donwload them and will instead flash them to the board using OpenOCD etc

bestware, 2023-11-04:
Same issue! 
But where to copy the hexfile to, to skip download by jar?

EmikVayts, 2023-11-10:
> Same issue! But where to copy the hexfile to, to skip download by jar?

I resolved the issue by manually downloading the 's110' hex file for example to this Windows directory (Not sure what the equivalent on Mac or Linux would be): 

C:\Users\YOURUSERHERE\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.7.0\cores\nRF5\SDK\components\softdevice\s110\hex

bestware, 2023-11-11:
This solved my issue. Thanks a lot!

bestware, 2023-11-16:
The hex-file, of course! Took exactly the same,
Did you get any error message?
Be aware on the Arduino-settings (Softdevice version, Board, Programmer...)

Bug38, 2023-11-16:
> The hex-file, of course!
Yeah I knew it, just to be sure. But anyways, both copies did the same.

> Did you get any error message?
Always the same message : 
`Downloading 'https://www.nordicsemi.com/-/media/Software-and-other-downloads/SoftDevices/S110/s110nrf51800.zip' ...
Error while flashing SoftDevice.
java.io.IOException: Server returned HTTP response code: 403 for URL: https://www.nordicsemi.com/-/media/Software-and-other-downloads/SoftDevices/S110/s110nrf51800.zip
`
I'm trying to use a NRF51 Dongle (PCA10031), I selected all the correct options, but...

Just to be sure, hex file is here :
"%AppData%\..\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.7.0\cores\nRF5\SDK\components\softdevice\s110\hex\s110_nrf51_8.0.0_softdevice.hex"

I tried s130 too

bestware, 2023-11-16:
I'm using ST-Link V2, but that's not the root cause.
I assume - as the script can't find the hex-file, but tries to download - there must be something wrong with the path. Suggest to verify current Windows user for the Appdata-path. No more ideas else...